wuxx/nanoDAP #6: Windows7 下无法识别问题

mengyyy, 2019-09-04:
win7 上插入dapLink 后 有时会出现安装驱动程序 但经常显示无法识别设备 

驱动问题显示为  **该设备已被禁用,原因是,设备的固件没有提供必需的资源.(代码29)**

后面查了一下 有人也遇到这个[问题](https://discuss.redbear.cc/t/daplink-v1-0-cannot-be-recognized-by-win7/1112/16) 

他的解决方法是先把daplink的一部分插入usb口  也就是让 usb的两个电源线先上电，再插入剩下的部分

实际上也能观察到 dapLink的usb接口处 边缘的两个触点是稍微长一点的 是可以做到让这两点先接触 再插入剩余部分让中间两点后接触的
```
Hi, I encounter the same issue. It seems the DAPLink power up time is too long. so when win7 tries to comunicate with it, the DAPLink is still in initialization, and so will not response. Then when its actually ready, win7 already terminates the connection.

Currently the only workaround i use is to plug the DAP link partially (to make sure its properly powered first) then fully push it to USB port to start conection.

PS : I also Tried in Mac and this issue did not exist. Seems like the driver in windows need some modification or the hardware need to be fixed.

Hope it helps.
```

wuxx, 2019-09-07:
你用的是什么硬件版本？是V2.0还是V2.1，V2.1已经在USB D+加上GPIO控制上拉了，这个问题应该不会出现了。

mengyyy, 2019-09-09:
v2.0

wuxx, 2019-09-09:
v2.0 D+上是直接上拉的，所以在部分windows系统上是会出现枚举失败的问题，慢慢插入可以解决，刷成cmsis-dap 的固件也可以解决。也可以尝试一下修改DAPLink启动代码

closed 2019-09-20 by wuxx