wuxx/nanoDAP #9: cannot load flash programming algorithm

Dachtire, 2020-07-05:
1. 用SONiX_USB_MCU_ISP_Tool往键盘刷了一个烂固件，导致键盘不能识别。由于键盘没有留swd接口，所以就把芯片吹下来接到烧录座上。
2. 想通过keil重新烧录固件，芯片是EVISION VS11K15A，通过QFP48-0.5烧录座引出swd接到daplink，keil用的是SONiX SN32F268F的cmsis和pack (VS11K15A没有给出官方测试工具，所以用SN32F268F的测试工具，目前看来这两个芯片是一样的，测试过用SONiX_USB_MCU_ISP_Tool_V2.3.1.7里SN32F268F的usb isp给VS11K15A写固件没问题)，网上搜到的降速之类的也没用。
3. 有恢复用的.hex固件，是否可以通过coflash之类的软件烧录？但是coflash在Flash Driver一项里点击Add就闪退。目前没有搜到coflash的教程，不知如何使用。
4. 新接触单片机不知道哪里做错了，搜索引擎什么有用的都搜不到，请求帮助。
5. 有关资料
SONiX SN32F268F
http://www.sonix.cn/article-cn-558-24749
[SN32F268_V1.2_SC.pdf](https://github.com/wuxx/nanoDAP/files/4874400/SN32F268_V1.2_SC.pdf)
[SN32F268_V1.9_EN.pdf](https://github.com/wuxx/nanoDAP/files/4874401/SN32F268_V1.9_EN.pdf)
EVISION VS11K15A
http://evision.net.cn/download/icsc.html
[VS11K15A-20190413174718_2058.pdf](https://github.com/wuxx/nanoDAP/files/4874395/VS11K15A-20190413174718_2058.pdf)
6.相关截图
[image]
[image]
[image]
[image]
[image]
[image]

wuxx, 2020-07-05:
keil烧录一般需要源码，直接烧录hex可以试试
https://github.com/wuxx/openocd-toolbox
需要自己参考写个配置脚本

Dachtire, 2020-07-07:
> keil烧录一般需要源码，直接烧录hex可以试试
keil用的是源码，来自sonix官方给的SN32F260_CMSIS Firmware Library_V1.2里的一个demo，还是想问问这个报错有没有办法解决。
> 需要自己参考写个配置脚本
openocd写好了配置脚本，但是flash driver有问题。
[test.cfg.txt](https://github.com/wuxx/nanoDAP/files/4885740/test.cfg.txt)

1. 如果用cfi，会报Segmentation fault无法写入。
cfg里写的`flash bank $_FLASHNAME cfi 0x00000000 $_FLASH_SIZE 4 4 $_TARGETNAME`
chip_width是32bit=4Byte, 但是不知道bus_width在ic手册里哪里可以找到？目前填的就是4Byte。

执行终端输出
sudo openocd -f test.cfg -c "program SN32F260-usb-demo.hex"
Open On-Chip Debugger 0.10.0+dev-01293-g7c88e76a-dirty (2020-07-07-01:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cortex_m reset_config sysresetreq

Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: FW Version = 0254
Info : CMSIS-DAP: Serial# = 07000001066fff373439575043063956a5a5a5a597969908
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : sn32f268f.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for sn32f268f.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x00000002 msp: 0x00905a4c
** Programming Started **
Segmentation fault

进telnet调试
> init
> halt
> reset init
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x00000002 msp: 0x00905a4c
> flash write_image erase SN32F260-usb-demo.hex
Connection closed by foreign host.

2. 如果尝试用其他如stm32f1x，就是无法识别。参考自https://github.com/smp4488/dk63/blob/master/vs11k09a-1.cfg
sudo openocd -f test.cfg -c "program SN32F260-usb-demo.hex"
Open On-Chip Debugger 0.10.0+dev-01293-g7c88e76a-dirty (2020-07-07-01:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
cortex_m reset_config sysresetreq

Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: FW Version = 0254
Info : CMSIS-DAP: Serial# = 07000001066fff373439575043063956a5a5a5a597969908
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : sn32f268f.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for sn32f268f.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x00000002 msp: 0x00905a4c
** Programming Started **
Info : device id = 0x00000000
Warn : Cannot identify target as a STM32 family.
Error: auto_probe failed
embedded:startup.tcl:521: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 586
at file "embedded:startup.tcl", line 521

3. 不知道和bootloader或者reset有没有关系，现在只在烧录座接了swd的四根线，其他针脚都是没接的。
ic说明书里描述的
BOOT — Internal pull-up in Boot loader, tie LOW to keep in Boot loader or left
HIGH to exit Boot loader and execute User program at boot time.
RESET — External Reset input. Schmitt trigger structure, active “Low”, normally
stay “High”.

wuxx, 2020-07-08:
先看keil中能否检测到芯片的idcode，这种错误有可能是芯片被保护了，导致无法烧录。

closed 2022-07-21 by wuxx