wuxx/nanoDAP #10: 编译的时候没有 RTL.h

xidianzhuxi, 2020-09-03:
编译的时候缺少rtl.h和usb_lib.c也没有core_cm3.h

xidianzhuxi, 2020-09-03:
上面问题解决了。现在是cmsis_compiler.h没有

wuxx, 2020-09-03:
参考
https://github.com/wuxx/nanoDAP/issues/3

xidianzhuxi, 2020-09-03:
这个问题我也解决了，现在的问题是这样的。我手上的这块是stm32F103RCT6.配置文件DAP_config.h需要如何修改呢？

wuxx, 2020-09-03:
应该不需要修改，直接编译出来就可以了

xidianzhuxi, 2020-09-03:
当然除了必要的GPIO口我更改了。有这么几个问题我没想明白：
1，firmware里面的hex文件是什么时候用
2，给MCU烧录的时候提示No Algorithm found for 08004000H-08008633H
然后虽然烧录成功了。但是检测不到,提示SWD/JTAG Communication Failure
另外关于LED灯的状态是什么意思呢，我有一个灯常亮，一个灯闪烁

xidianzhuxi, 2020-09-03:
ok，查看了DAP发现 PA0 是SWDIO，PA4是SWCLK。而不是电路图上的P12，P13.现在检测到了

xidianzhuxi, 2020-09-03:
烧录的时候一直是RDDI-DAP ERROR 速度我挨个都试了一遍都没有用。有没有可能是芯片不支持，或者是烧录算法的缺失？

wuxx, 2020-09-03:
这个是检测不到目标芯片，最常见的原因：
1. 线没有接好
2. 目标芯片没有上电

对于你从源码编译的固件，可能是IO没有配置对，参考一下提供的v2.3原理图

xidianzhuxi, 2020-09-04:
今天拿了宁一块ministm32 也就是stm32RCT6，发现烧录成功了。昨天那块stm32VB是检测不到目标芯片。根据文档说DAPLINK支持的芯片列表有限，所以我如果要自己针对其他芯片支持我需要怎么做？

**您可以自行编译固件，或者从本仓库的firmware目录下直接获取即可，假若您有自己的芯片平台需要支持拖拽烧录，可以参考目前的代码作修改。**

我比较困惑的是
1我似乎没有用到firmware目录下的文件.
2参考哪些代码做修改？
3修改后的代码该怎么用
感谢

wuxx, 2020-09-04:
1. 如果参考提供的原理图制作的DAP，可以直接烧录firmware下的固件
2. 如果用keil开发，基本上可以烧录所有cortex-m系列的单片机
3. 拖拽烧录每种固件一般只支持一种单片机烧录，支持其他单片机拖拽烧录需要更换固件，若DAPLink官方没有支持，则需要自己修改代码支持拖拽烧录其他目标芯片，具体参考flash_blob.c源文件，修改好重新编译就可以了

closed 2020-09-05 by wuxx