wuxx/nanoDAP #17: Why try to read a output pin?

llinjupt, 2021-09-04:
Hi wu,

I am not sure if it is an issue but it seems really strange.

JTAG TDI has been configured as an out put pin through pin_out_init().
At the same time there is another function PIN_TDI_IN() that tries to read this pin.

PIN_TDI_IN() is only called by DAP_SWJ_Pins() to check if the electrical level of TDI is changed.
So my question is that do you have any idea of this contradictory opeation -- try to read an output pin.

Thanks in adavnce!

wuxx, 2021-09-06:
Hello, I think it's just a convention,  the DAP_SWJ_Pins read all pin input value, include the TCK/SWCLK, nRESET, their are all config as output. the host application know it.

closed 2021-09-06 by wuxx

llinjupt, 2021-09-06:
> 
> 
> Hello, I think it's just a convention, the DAP_SWJ_Pins read all pin input value, include the TCK/SWCLK, nRESET, their are all config as output. the host application know it.

Thanks for your response and you are right that all these pins configured at output mode are read in DAP_SWJ_Pins(). I have checked NXP LPCxx MCU and STM32 specs which truly are able to support this freature. But I am pretty sure there are lots of MCUs that don't support such a feature to read an output pin. I am gonna check JTAG spec to see if it is a must be, maybe it is  a real issue. Thanks again.