wuxx/nanoDAP #19: Linux中使用openocd/cmsis-dap返回错误

btvnlue, 2022-04-01:
你好，使用Linux的openocd 0.11.0-rc2链接cmsis-dap的interface，返回
Open On-Chip Debugger 0.11.0-rc2
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Using CMSIS-DAPv2 interface 1 with wrong class 10 subclass 0 or protocol 0
Info : Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf001, serial=0001A0000001
Warn : could not claim interface: Resource busy
Error: error writing data: Input/Output Error
Error: CMSIS-DAP command CMD_INFO failed.

试着刷了2.3最新的固件，结果还是一样，
有没有什么建议

btvnlue, 2022-04-01:
找到了，不好意思，要用v1协议，加入command
cmsis_dap_backend hid
就可以了

AIMERSAN, 2022-07-02:
你好 请问是如何在终端添加这条命令（cmsis_dap_backend hid）呢， 能否方便复制下

wuxx, 2022-07-02:
tcl/interface/cmsis-dap.cfg

AIMERSAN, 2022-07-02:
> tcl/interface/cmsis-dap.cfg
wuxx 谢谢你回复 不过我说的是openocd 的运行选项，你这个说的应该是openocd的启动配置文件吧。
另外这个问题我也解决了正如题主说的那样，只需要在启动命令添加cmsis_dap_backend hid即可，命令如下：openocd -f ~/自己的配置路径/openocd.cfg -c "cmsis_dap_backend hid"

LiHangBing, 2023-05-31:
不知道什么原因，换了个USB口（之前插在HUB上），居然能用了

wuxx, 2023-06-01:
好的

closed 2023-06-01 by wuxx