wuxx/nanoDAP #23: 支不支持用nanoDAP JTAG调试VisionFive开发板？

worksage, 2022-07-02:
您好老师：@wuxx， 我从资料上理解到可以用nanoDAP JTAG调试RISC-V SOC，我的开发板是VisionFive开发板(https://github.com/starfive-tech/VisionFive)，但我之前没用过JTAG，所以需要弄明白搞确切！！！
https://forum.rvspace.org/t/connecting-to-visionfive-s-jtag-port-a-short-guide/514
上面是请教过的内容，还有这一篇关于VisionFive开发板的JTAG详情的文章！！！

想请老师给我些用nanoDAP JTAG调试RISC-V SOC的指导！！！如果适用VisionFive购买哪款nanoDAP JTAG更合适？？？

wuxx, 2022-07-02:
你好，应该是用openocd调试对吗，可以的

wuxx, 2022-07-02:
一般情况推荐使用DAPLink，是最新的开源DAPLink代码，使用上可能更方便

worksage, 2022-07-02:
> 

是的，ubuntu下openocd + gdb，还差仿真器没建起来

worksage, 2022-07-02:
我刚看老师的视频了，选那个高速的DAPLink更好是吧！如果用着合适，我会推荐到下面的论坛里：
https://forum.rvspace.org/t/connecting-to-visionfive-s-jtag-port-a-short-guide/514

wuxx, 2022-07-02:
好的，高速的速度更快一些，日常使用用普通速度的一般也够了

worksage, 2022-07-02:
还有请问老师，我不打算焊接JTAG插接头，怎么接线更好，下面是图片https://forum.rvspace.org/uploads/default/original/1X/db9e6878a00f6c3ca5b3cafbfd744d82c5070c1d.jpeg，

wuxx, 2022-07-02:
一般用杜邦线连接即可

worksage, 2022-07-02:
老师，若不焊接，那七个通孔怎么接杜邦线？

wuxx, 2022-07-02:
https://github.com/wuxx/nanoDAP/wiki/%E5%85%B3%E4%BA%8E%E6%8E%A5%E7%BA%BF%E8%AF%B4%E6%98%8E

worksage, 2022-07-02:
https://item.taobao.com/item.htm?spm=a1z10.5-c.w4002-21349689069.20.2b254af5TLGSqz&id=605653531236
老师，这款也可作为USB转TTL使用是吧

wuxx, 2022-07-02:
可以

worksage, 2022-07-02:
老师，我看上面的链接也没看明白https://github.com/wuxx/nanoDAP/wiki/%E5%85%B3%E4%BA%8E%E6%8E%A5%E7%BA%BF%E8%AF%B4%E6%98%8E，VisionFive开发板上那标着A的7个通孔必须焊接上插针吗？有没有免焊接的插针座？

wuxx, 2022-07-02:
需要焊插针

wuxx, 2022-07-02:
如果不焊插针，可以看看这种夹子行不行
[image]

worksage, 2022-07-02:
感谢老师解惑！！！

closed 2022-07-09 by wuxx