wuxx/nanoDAP #27: 不识别芯片怎么回事啊

bihailantian655, 2023-04-27:
我用的103cb开发板，刷的固件，想测试一下能不能用。
[image]
这俩100r 的电阻没接直接接的目标板。
keil中能识别 daplink ，刷的是2.3的这俩固件测试都不行。
[image]
[image]

under reset 和 norml 都不行
[image]

bihailantian655, 2023-04-27:
开发板原来的pb12 pb13引脚有外围电路，更改引脚为pb10 pb11以后。
出现了一个现象就是
keil 连接4个jtag 引脚 能用swd 模式识别芯片， jtag模式不能识别芯片
 只连接swdio swclk  gnd vcc    swd模式 不能识别被烧录的芯片。

LiHangBing, 2023-06-01:
[RadioOperator](https://github.com/RadioOperator)
/
[STM32F103C8T6_CMSIS-DAP_SWO](https://github.com/RadioOperator/STM32F103C8T6_CMSIS-DAP_SWO)
这个里面的蓝色药丸remap固件实测可以识别

closed 2023-12-13 by wuxx