wuxx/nanoDAP #28: NXP MCUXpresso IDE 无法连接

MrAMS, 2023-05-25:
使用NXP MCUXpresso IDE 的 LinkServer连接RT1064板子时，IDE报错：

[image]

详细信息：

> Error in final launch sequence:
> 
> Failed to execute MI command:
> -target-select extended-remote localhost:10989
> Error message from debugger back end:
> Remote communication error.  Target disconnected.: (undocumented errno 10061).
> Failed to execute MI command:
> -target-select extended-remote localhost:10989
> Error message from debugger back end:
> Remote communication error.  Target disconnected.: (undocumented errno 10061).
>

MrAMS, 2023-05-25:
已排查接线问题，
NXP LinkServer配置如下：
[image]
[image]

wuxx, 2023-05-26:
先试试看用有线的可以吗

MrAMS, 2023-05-26:
已经识别上了而且LinkServer连接上了
[image]
身边暂时没有其他的DAP

closed 2024-07-01 by wuxx