wuxx/nanoDAP #31: SWD multidrop support?

arthurfprecht, 2023-10-02:
Hello!

I first would like to say that I appreciate this debugger very much. As far as my research goes, it is the only DAP debugger with JTAG and SWD support that runs on STM32F103 MCU, which is very handy (it is a very common MCU). 
With the arrival of RP2040, the SWD support of the current nanoDAP is not up to date - RP2040 uses multidrop  SWD. I know that the DAPLink [mBed repo](https://github.com/ARMmbed/DAPLink) has an updated version of the firmware for STM32F103 - but then it does not support JTAG. 

Would it be possible to add SWD multidrop support to the current nanoDAP? or, on the other hand, add JTAG to the newer DAP firmware, so we get the best of both worlds (JTAG support and multidrop SWD support, so ESP32, RP2040 and other targets can all be debugged by a single debugger, on STM32F103)? 

Thank you for the attention!

wuxx, 2023-10-03:
[flash_image_for_pico.zip](https://github.com/wuxx/nanoDAP/files/12787622/flash_image_for_pico.zip)

wuxx, 2023-10-03:
hello，this 128KB firmware support multi-drop SWD（support RP2040）thanks~！
you can also check this repo  https://github.com/wuxx/pico-lab

arthurfprecht, 2023-10-03:
Woow, that's amazing! I will test it today! Thank you very much for the info and the image!

closed 2023-10-06 by wuxx