wuxx/nanoDAP #37: 用 nano-DAP 來刷新 PICO、PICOW、PICO2 和 PICO2W (SWDIO)

symdeb, 2025-02-08:
我想使用 nano-DAP 來刷新 PICO、PICOW、PICO2 和 PICO2W (SWDIO)

將 GND、RESET、TCL、TCK 線連接至 PICO

Bus 002 Device 028: ID 0d28:0204 NXP ARM mbed

我可以看到 nanoDAP 的用途STM32F103CBT6. 所以看起来这是一个 DAP-LINK

openocd -f interface/cmsis-dap.cfg -f target/rp2350-dbgkey-nonsecure.cfg -c "adapter speed 5000" -c "main.elf verify reset exit"

[1] ]在哪裡可以找到 nano-DAP 的 cmis-dap.cfg？
[2] 使用來自 https://github.com/raspberrypi/openocd/tree/sdk-2.0.0/tcl/target 的 cfg 檔案（請參閱附件）

[cfg.zip](https://github.com/user-attachments/files/18636216/cfg.zip)

Open On-Chip Debugger 0.12.0+dev-gebec950 (2025-01-20-23:51)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Error: unable to find a matching CMSIS-DAP device

Nano-DAP 可以與 PICO 一起使用嗎？ 如果是，那麼可能是什麼問題？

wuxx, 2025-02-13:
https://github.com/wuxx/pico-lab

symdeb, 2025-02-15:
TMS_IO
 GND 
 TCK 

1. Bus 001 Device 004: ID 0d28:0204 NXP ARM mbed

2. sudo dmesg
[5192.482050] cdc_acm 2-1.2.1:1.0: ttyACM0: USB ACM device
[ 5210.173463] usb 1-2.2: new full-speed USB device number 4 using xhci_hcd
[ 5210.269168] usb 1-2.2: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[ 5210.269176] usb 1-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5210.269179] usb 1-2.2: Product: DAPLink CMSIS-DAP
[ 5210.269181] usb 1-2.2: Manufacturer: ARM
[ 5210.269183] usb 1-2.2: SerialNumber: 070000010668ff303430484257205021a5a5a5a597969908
[ 5210.280576] cdc_acm 1-2.2:1.1: ttyACM1: USB ACM device
[ 5210.285113] hid-generic 0003:0D28:0204.0003: hiddev1,hidraw2: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-0000:04:00.0-2.2/input3
[ 5210.336976] usb-storage 1-2.2:1.0: USB Mass Storage device detected
[ 5210.340175] scsi host8: usb-storage 1-2.2:1.0
[ 5210.340336] usbcore: registered new interface driver usb-storage
[ 5210.345511] usbcore: registered new interface driver uas
[ 5211.380033] scsi 8:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
[ 5211.380423] sd 8:0:0:0: Attached scsi generic sg5 type 0
[ 5211.382243] sd 8:0:0:0: [sde] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
[ 5211.382601] sd 8:0:0:0: [sde] Write Protect is off
[ 5211.382604] sd 8:0:0:0: [sde] Mode Sense: 03 00 00 00
[ 5211.382827] sd 8:0:0:0: [sde] No Caching mode page found
[ 5211.382829] sd 8:0:0:0: [sde] Assuming drive cache: write through
[ 5211.397451]  sde:
[ 5211.397507] sd 8:0:0:0: [sde] Attached SCSI removable disk

3.  pull openocd
     $./configure --enable-cmsis-dap       
    make

4. Add 2 lines to cmsis-dap.cfg   
    cmsis_dap_backend hid   
     transport select swd

5, sudo openocd -f interface/cmsis-dap.cfg -f target/rp2350.cfg 

Open On-Chip Debugger 0.12.0+dev-gebec950 (2025-01-20-23:51)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Hardware thread awareness created
cortex_m reset_config sysresetreq
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Error: unable to find a matching CMSIS-DAP device

6, $openocd -f interface/cmsis-dap.cfg -f target/rp2350-dbgkey-nonsecure.cfg -c "adapter speed 5000" -c "main.elf verify reset exit"
 
 7.  DEPRECATED! use 'cmsis-dap backend', not 'cmsis_dap_backend' 

8. sudo openocd -f interface/cmsis-dap.cfg -f target/rp2350.cfg 

Open On-Chip Debugger 0.12.0+dev-gcf9c0b41c (2025-02-15-08:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
invalid backend argument to cmsis-dap backend <backend>

cat ./tcl/interface/cmsis-dap.cfg
# SPDX-License-Identifier: GPL-2.0-or-later

#
# ARM CMSIS-DAP compliant adapter
#
# http://www.keil.com/support/man/docs/dapdebug/
#

adapter driver cmsis-dap

# Optionally specify the serial number of CMSIS-DAP usb device.
# adapter serial 02200201E6661E601B98E3B9
cmsis-dap backend hid

symdeb, 2025-02-15:
PICO using CMSIS DAP interface
---------------------------------
```
git clone --recurse-submodules --remote-submodules https://github.com/openocd-org/openocd.git 
cd jimtcl
./configure
make
sudo make install
cd ..

git clone https://github.com/libusb/hidapi.git
git clone hidapi
cd hidapi
./bootstrap
.configure
make
sudo make install

./bootstrap
./configure --enable-cmsis-dap
make
sudo make install
sudo openocd -f interface/cmsis-dap.cfg -f target/rp2350.cfg -c "program <project>.elf verify reset exit"

sudo openocd -f interface/cmsis-dap.cfg -f target/rp2350.cfg -c "program test_project.elf verify reset exit"
Open On-Chip Debugger 0.12.0+dev-01855-g73e9b7898-dirty (2025-02-15-22:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : [(null)] Hardware thread awareness created
Info : [(null)] Hardware thread awareness created
Info : [rp2350.dap.core0] Hardware thread awareness created
Info : [rp2350.dap.core1] Hardware thread awareness created
cortex_m reset_config sysresetreq
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 0254
Info : CMSIS-DAP: Serial# = 070000010668ff303430484257205021a5a5a5a597969908
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core0] target has 8 breakpoints, 4 watchpoints
Info : [rp2350.dap.core0] Examination succeed
Info : [rp2350.dap.core1] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core1] target has 8 breakpoints, 4 watchpoints
Info : [rp2350.dap.core1] Examination succeed
Info : [rp2350.dap.core0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Warn : [rp2350.dap.core1] VECTRESET is not supported on this Cortex-M core, using SYSRESETREQ instead.
Warn : [rp2350.dap.core1] Set 'cortex_m reset_config sysresetreq'.
[rp2350.dap.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf9000000 pc: 0x00000088 msp: 0xf0000000
[rp2350.dap.core1] halted due to breakpoint, current mode: Thread 
xPSR: 0xf9000000 pc: 0x00000088 msp: 0xf0000000
** Programming Started **
Error: couldn't open test_project.elf
** Programming Failed **
shutdown command invoked

```

symdeb, 2025-02-15:
PICO using CMSIS DAP interface (SOLVED)
=========================================================================================
```
git clone --recurse-submodules --remote-submodules **https://github.com/raspberrypi/openocd.git** 

# when using raspberry git repo instead of openOCD git
# to skip calloc error
**export CFLAGS="-Wno-error=calloc-transposed-args"**

./bootstrap
./configure --enable-cmsis-dap
make
sudo make install

sudo openocd -f interface/cmsis-dap.cfg -f target/rp2350.cfg -c "program pico-ssd1306.elf verify reset exit"

```

closed 2025-04-16 by wuxx